rkistner/arcore #18: Arduino v1.6.3 Board Manager Package

toddtreece, 2015-04-06:
I have been working on adding support for Adafruit's boards to the new Arduino IDE Board Manager, and I packaged up your core and added it to our proxy server. The proxy server injects new boards into the Board Manager's `package_list.json` file and serves the modified version back to the IDE. More info can be found here: https://blog.adafruit.com/2015/04/06/adding-custom-boards-to-the-arduino-v1-6-board-manager/

Not sure if that's helpful, but it might be easier than rebuilding the IDE for new releases. Let me know if you have any questions, or if you would like me to remove it from the proxy.

NicoHood, 2015-04-06:
I've also added the MIDI functions to my extended HID core. Just talked with Ralf today. I need people to test this. If you have a chance, let me know. It works the same as his version, just select any USB core. It is currently under development, a proper IDE entry will be added if it works:
https://github.com/NicoHood/HID/tree/Midi-Test

It would also be nice if you could add my core as well (master).

toddtreece, 2015-04-06:
@NicoHood if you want to package up your stuff and submit a pull req, we should be able to add it. check out this https://github.com/esp8266/Arduino/issues/57 and the readme in the proxy repo for more info.

rkistner, 2015-04-06:
Thanks @toddtreece. I haven't had a chance to look at Arduino 1.6 yet. Does it just work, or is there anything I should do from my side?

toddtreece, 2015-04-07:
@rkistner I have a fork that changes the structure a bit because of the way packages are made for the board manager, but the core itself didn't change other than some minor modifications. I removed support for the boards I couldn't test and confirm working on the proxy version, so only the Micro and Leonardo are present.

Here's my fork, but it probably shouldn't be merged: https://github.com/toddtreece/arcore It might make sense eventually once the board manager stabilizes.

stefandz, 2015-07-15:
Sorry to jump in on an old issue here chaps, but this seems the right place to put this. @rkistner and @toddtreece - I work for Bare Conductive and contributed the iPad reported current tweak and support for the Touch Board. We're not planning on using the Board Manager for our USB MIDI support, but do want to be able to distribute this code (for free, obviously) to our customers.

@toddtreece - this means that we won't be asking to be included in your fork, and you can keep it for boards you have access to test

@rkistner - in order to keep things clean and simple for our users, I would like to include arcore in https://github.com/BareConductive/bare-conductive-arduino. This won't be a direct fork, but the code will be there (and credited however you like best - I was going to put a link in readme.md). Is this acceptable? Obviously any changes I make I will make known here, and likewise any improvements to this repo I shall manually port across.

If anyone thinks this might not be such a great idea, please let me know.

rkistner, 2015-07-15:
@stefandz Sure, I don't have any problem if you copy the code. Let me know if I need to change anything here to help you.

I still need to look at the new board manager, but I haven't had the time to work on this in a while.

stefandz, 2015-07-15:
Cool, thanks @rkistner - the only necessary changes will be removing Touch Board support (to avoid confusion). I'll submit these as a pull request once we have our end tied up.

NicoHood, 2015-07-16:
Midi USB is now in the current hourly IDE build:
https://github.com/arduino/Arduino/wiki/PluggableUSB-and-PluggableHID-howto
https://github.com/arduino/Arduino/pull/3304

stefandz, 2015-07-21:
Thanks for letting us know, @NicoHood - do you happen to know if 1.6.6 has a solid release date yet?

NicoHood, 2015-07-21:
I dont know what they plan. Maybe its on google code.
I tested the code today and submitted some changes. Its possible that they release it soon. It'd be better if people test the new USB Core before. You can download the nightly build.