rkistner/arcore #19: How to use pitch bend ?

orgicus, 2015-12-22:
Hi,

I'm new to MIDI and getting a noteOn/noteOff with arcore + Leonardo was pretty easy.
This isn't an issue per se, but could please explain and point me in the direction on how to send pitch bend data (maybe on two different channels) ?

Thank you,
George

rkistner, 2015-12-29:
Did you get something working yet? In the MIDI-USB spec there are always 4 bytes per event. It typically adds a byte to the start of the normal MIDI message. This is just the event type, `0x0E` in this case.

You can try the following code (haven't tested it myself):

```
// The pitch bend value is a 14-bit number (0-16383). 0x2000 (8192) is the default / middle value.

// First byte is the event type (0x0E = pitch bend change).
// Second byte is the event type, combined with the channel.
// Third byte is the 7 least significant bits of the value.
// Fourth byte is the 7 most significant bits of the value.
void pitchBendChange(byte channel, int value) {
  byte lowValue = value & 0x7F;
  byte highValue = value >> 7;
  MIDIEvent event = {0x0E, 0xE0 | channel, lowValue, highValue};
  MIDIUSB.write(event);
}
```

orgicus, 2016-01-02:
Awesome! This worked like a charm.
Should've read the specs carefully.
Sorry about the late reply and thank you very much.

closed 2016-01-02 by orgicus

rkistner, 2016-01-04:
Great, glad it worked!