rkistner/arcore #21: midi over USB + midi over DIN - how to use both inputs and outputs?

hackspacecatania, 2016-04-28:
i'm working on a new opensource project based on arcore/midi and Leonardo boards. 
when i set the baudrate to the classic 31250 i can send midi data over midi din cables (using serial1) and over arcore/midi via usb (MIDIUSB.write) ...
there are many problems regarding the incoming midi messages! the datas received via usb are not precise at all!
same problems if i set the baud rate at common speeds like 57600 or 115200 and so on... 

if i do not set the serial.begin baudrate the usb comunication is perfect for midi in and out... but i can't have the classic midi over din :(

NicoHood, 2016-04-28:
Try this lib:
https://github.com/arduino-libraries/MIDIUSB

rkistner, 2016-05-03:
@hackspacecatania There may be something in the Serial code that's affecting the USB stack. This project is based on quite an old version of the Arduino USB stack, and newer versions may work better (and have probably changed substantially).

Thanks for the link @NicoHood. Please let me know how that works - if it is working well I will deprecate this project.

hackspacecatania, 2016-05-03:
i tested the library that NicoHood linked, but with no success al all! 
all the midi software on my computer crashed when receiving midi data... 
this is the project i'm working on : [https://www.facebook.com/DARTcontroller/?ref=ts&fref=ts](url)

Using Arcore midi everithing works good, except that if i specify Serial.begin(31250) the USB-IN messages are corrupted... i'll try with newer arduino ide's
i'll also try to contact the arduino.org team

NicoHood, 2016-05-03:
If it does not work, file a bug there. I have not tried this code a single time, and I have nothing to do with it. Just wanted to give you another option.

hackspacecatania, 2016-05-03:
Sure! And thank you for doing that :) i Just did a quick try with MIDIUSB.  May be i made some mistakes... I'll  report my experience here first, anyway