rkistner/arcore #26: How to change the name of the device?

silveirago, 2017-06-26:
I'm trying to change the name in boards: micro.build.usb_product="Arduino Micro” to micro.build.usb_product="other name”, but it doesn't change anything, it still shows up as Arduino Micro. Is there a way to change its name?

Thanks!

rajivdeo, 2017-08-08:
Looking for a easy way for the burning issue without creating a device id and manufacturer registration. Teensy provides the functionality to the newbie for changing the USB Midi name in Windows. Hence, using ARCORE should also be feasible. Please share the mechanism.

silveirago, 2017-08-08:
I found a way. Go to Aplications > Arduino (where it's installed Arcore)
right click > show package contents > Java > Hardware > Arduino > avr >
boards.txt
Save this file in a another place so you don't lose it. Find something like
it:
micro.build.vid=0x2341
micro.build.pid=0x8037
micro.build.usb_product="Arduino Micro"

And change for something like it:

micro.build.vid=0x999
micro.build.pid=0x999
micro.build.usb_product="My midi controller"
And save the file.

I'm saying this by memory, because I don't have my files with me now, but
it's really something like that!

Ps: I'm not using Arcore Anymore, I'm using the MidiUSB library, it works
similar but works better. But, Arcore is great, though.

rajivdeo, 2017-08-09:
Thanks silveirago for your suggestion. I tried this and now I am getting "USB IO board" instead of "Arduino Micro" as the device description. Still far away from desired goal.
My intent is to use Arduino Micro Pro for custom midi controller with custom name. If you are currently using MidiUSB library you can even let me know how to solve the problem using that library.

starock, 2017-10-20:
Still doesnt work~ can't change the usb device name.

> micro.build.vid=0x999
> micro.build.pid=0x999
> micro.build.usb_product="My midi controller"
> And save the file.

rajivdeo, 2017-10-20:
Still can't change the usb device name to anything other than "USB IO Board". Please help. This feature is badly needed.

silveirago, 2017-10-24:
micro.build.vid=0x999
micro.build.pid=0x999
Should be different numbers.

Try:
micro.build.vid=0x998
micro.build.pid=0x999

rajivdeo, 2017-10-24:
Thanks for quickly responding. Would this work for arcore?

silveirago, 2017-10-24:
I use it for the midiusb library. I’m not sure about the Arcore. The usb library adds on to the arduino micro board that is already installed in the arduino folder, while the Arcore adds a “new board”. I believe that for changing the board name in the Arcore you should open the txt file that’s in the Arcore folder and not the one that comes with the arduino. If this makes sense...

rajivdeo, 2017-11-04:
Not working after editing boards.txt for arcore.
Any further help?

silveirago, 2017-11-04:
I'm trying to help you with Arcore, but I just don't use Arcore anymore,
just the usbMidi library. And it works fine. Both are good, but I think I
can't help you with Arcore. :/ I would suggest trying the other one.

rajivdeo, 2017-11-04:
Thanks Silveirago for all the help. 
Can you suggest what changes I need to make in order to change from arcore to midiusb?  Please confirm whether I am on the right path -
1. change board in Arduino IDE from pro micro (arcore) to sparkfun pro micro.
2. I assume that non-midi code remains untouched. Will it work without any changes?
3. Include midiusb.h ?
I am not sure what changes to the midi specific code I need to make. Any changes expected in latency after shifting to midiusb library? Any significant change in the code size / footprint?

silveirago, 2017-11-05:
You are right in those steps. You just need to take a look in the reference about the sintax of the midi message.

musinou, 2018-01-19:
Hi, I was wondering if someone here succeeded changing MIDI device's name. 

I use MIDIUSB.h or midi_UsbTransport.h
So, I added this board, and I still have "Arduino Leonardo" as MIDI device name. 
Any ideas?

Thank you!

leonardo2.name=Arduino Leonardo name

leonardo2.vid.0=0x999
leonardo2.pid.0=0x998
leonardo2.vid.1=0x997
leonardo2.pid.1=0x996
leonardo2.vid.2=0x995
leonardo2.pid.2=0x994
leonardo2.vid.3=0x993
leonardo2.pid.3=0x992
        
leonardo2.upload.tool=arduino:avrdude
leonardo2.upload.protocol=avr109
leonardo2.upload.maximum_size=28672
leonardo2.upload.maximum_data_size=2560
leonardo2.upload.speed=57600
leonardo2.upload.disable_flushing=true
leonardo2.upload.use_1200bps_touch=true
leonardo2.upload.wait_for_upload_port=true
        
leonardo2.bootloader.tool=arduino:avrdude
leonardo2.bootloader.low_fuses=0xff
leonardo2.bootloader.high_fuses=0xd8
leonardo2.bootloader.extended_fuses=0xcb
leonardo2.bootloader.file=caterina/Caterina-Leonardo.hex
leonardo2.bootloader.unlock_bits=0x3F
leonardo2.bootloader.lock_bits=0x2F
        
leonardo2.build.mcu=atmega32u4
leonardo2.build.f_cpu=16000000L
leonardo2.build.vid=0x2341
leonardo2.build.pid=0x8036
leonardo2.build.usb_product="Arduino Leonardo product"
leonardo2.build.usb_manufacturer="Arduino Leonardo manifacturer"
leonardo2.build.board=AVR_LEONARDO
leonardo2.build.core=arduino:arduino
leonardo2.build.variant=arduino:leonardo
leonardo2.build.extra_flags={build.usb_flags}

silveirago, 2018-01-19:
Yes. But, I’m using the MIDIUSB library instead. I made a full tutorial on that. The third video is where I show how you can change the name.

musinou, 2018-01-20:
Thank you @silveirago ! This helped. I liked the way @facchinm, in another thread, added a new board, so one would not have to change the Arduino app itself. This is how I ended up doing it: http://liveelectronics.musinou.net/MIDIdeviceName.php