rkistner/arcore #30: Midi to PWM

fvsr, 2018-01-16:
I was working on a code which transfers midi values to a PWM output of the Arduino Pro Micro. It worked but the maximum PWM frequency I was able to get was 1KHz. I made some tests and I wonder if the Arcore code slows down the clock frequency (suposed to be 16MHz). I made a similar code with SPI and I didnt have any problems. Any information about the frequency of operation of the Arcore code at the Arduino Pro Micro would be nice, thank you.

fvsr, 2018-01-17:
P.S.: I am talking about hardware PWM.

rkistner, 2018-01-17:
This is effectively a fork of an old version of the standard Arduino firmware. Only the USB stack was modified, and that shouldn't affect the clock frequency. Unfortunately I don't know much about PWM with Arduino, so can't really help you there.