rkistner/arcore #31: How to study the USB stack

fvsr, 2018-03-11:
I would like to dig deeper in the Arcore code, so would you give me any sugestion from wich files should I begin my study? We have the following path: "arcore-master/hardware/avr/cores/arcore", where there are 40 files. All of them are sheets or parts of the Arcore code? And how about the folder "libraries"? Would there be an order to best study and understand all those files, in order to better understand USB over MIDI? Thanks.

rkistner, 2018-03-12:
The USB code specific to MIDI (this project) is in these two new files:

https://github.com/rkistner/arcore/blob/master/hardware/avr/cores/arcore/MIDIUSB.cpp
https://github.com/rkistner/arcore/blob/master/hardware/avr/cores/arcore/MIDIUSB.h

As well as changes in these files:

https://github.com/rkistner/arcore/blob/master/hardware/avr/cores/arcore/USBAPI.h
https://github.com/rkistner/arcore/blob/master/hardware/avr/cores/arcore/USBDesc.h
https://github.com/rkistner/arcore/blob/master/hardware/avr/cores/arcore/USBCore.cpp
https://github.com/rkistner/arcore/blob/master/hardware/avr/cores/arcore/USBCore.h

Nothing is implemented as a library.

You can see all the changes here:
https://github.com/rkistner/arcore/compare/05fb7926f8d871ef5dc28055964d36a2dc8ab12f...master

This project is a fork of a very old version of the core Arduino firmware. Apparently newer versions allow "pluggable" USB libraries, that don't require modifying the core firmware like this. There may even be an existing USB-MIDI library for it.

In this project the USB implementation is based on the serial-over-USB implementation of the core firmware, since it functionally works in a very similar way. The most significant differences are that messages are 4 bytes each (instead of a stream of single bytes), and the USB descriptors are way more complicated. I also have to admit that my knowledge of USB development is very limited, so the code in this project may not be the best examples to learn from.

If you want to study the USB stack I also suggest that you familiarise yourself with all the tools available, and I strongly suggest using either Linux or OSX (not Windows):
https://github.com/rkistner/arcore/blob/master/USB-dev-notes.md

fvsr, 2018-03-12:
Thank you very much.